ComPair/ComPair-tracker-FPGA #43: Reading livetime counters doesn't return the same value even when ASIC should be sleeping.

SeanCGriffin, 2020-02-21:
Probably related to the registers being read at slightly different times; we'll need to figure out how to correct for this.
[image]

lucasparker, 2020-03-30:
I am going to close this issue -- now the real-time and live-time counters are included in the asic data packets, and the values get latched at the same instant

closed 2020-03-30 by lucasparker

adamschoenwald, 2020-09-12:
Does the act of checking the counter stop the live counter?

SeanCGriffin, 2020-09-14:
I believe that they are mapped to slv_regN's, so if they're free-running they shouldn't be affected by a read, since everything is synchronous to the AXI clk. Maybe @lucasparker can comment.