ComPair/ComPair-tracker-FPGA #113: Event_ID_Latch clock buffer placement error

adamschoenwald, 2020-09-14:
The event_id_latch input is not on a clock capable input in REVA (not sure, may have been in REV-). I think that is why i get the following error trying to compile the REVA firmware:
```
Place 30-574] Poor placement for routing between an IO pin and BUFG. If this sub optimal condition is acceptable for this design, you may use the CLOCK_DEDICATED_ROUTE constraint in the .xdc file to demote this message to a WARNING. However, the use of this override is highly discouraged. These examples can be used directly in the .xdc file to override this clock rule.
	< set_property CLOCK_DEDICATED_ROUTE FALSE [get_nets EVENT_ID_LATCH_IN_IBUF] >
	EVENT_ID_LATCH_IN_IBUF_inst (IBUF.O) is locked to IOB_X1Y64
	 and EVENT_ID_LATCH_IN_IBUF_BUFG_inst (BUFG.I) is provisionally placed by clockplacer on BUFGCTRL_X0Y31
```

I think we need to either add the constraint they say not to, or re-write the hdl at https://github.com/ComPair/ComPair-tracker-FPGA/blob/master/cores/vata_460p3/src/event_id_s2p.vhd

I think the problem is further compounded by including the event_id_s2p block in every vata core. It may be necessary to ensure this is being optimized to a single instance by either moving the code out of each block into its own single block which then fans out an event id bus, or at least crosses the clock domain in one location and not 12.

adamschoenwald, 2020-09-14:
I think this might go away if we trigger off of a rising edge detector instead of the rising_edge() macro but I'm not positive.

adamschoenwald, 2020-09-15:
make compile has no errors when I swap the constraints pad assignment of event_id_latch with a clock capable pin, so this is indeed the problem. 

I think we should implement a proper CDC, we may need to consult with someone on doings so I'm no expert on them.

lucasparker, 2020-09-18:
I am attempting a fix, which performs edge detection on event_id_latch in event_id_s2p... hopefully this is good enough?

lucasparker, 2020-09-18:
I am going to close this issue, since hopefully it is fixed with merging branch issue-113 into master

closed 2020-09-18 by lucasparker