ComPair/ComPair-tracker-FPGA #131: Only some ASICS read out on REVA

adamschoenwald, 2020-12-01:
Attempt to keep track of work done to fix this issue

adamschoenwald, 2020-12-01:
Post Swap configuration register readout:
ASIC 0
----------------
Setting config; response:  ok
Setting up trigger configuration: 
	 syncctrl forced_readout triggers: ok
ASIC 1
----------------
Setting config; response:  ok
Setting up trigger configuration: 
	 syncctrl forced_readout triggers: ok
ASIC 2
----------------
Setting config; response:  error: config_check
Setting up trigger configuration: 
	 syncctrl forced_readout triggers: ok
ASIC 3
----------------
Setting config; response:  error: config_check
Setting up trigger configuration: 
	 syncctrl forced_readout triggers: ok
ASIC 4
----------------
Setting config; response:  ok
Setting up trigger configuration: 
	 syncctrl forced_readout triggers: ok
ASIC 5
----------------
Setting config; response:  ok
Setting up trigger configuration: 
	 syncctrl forced_readout triggers: ok
ASIC 6
----------------
Setting config; response:  ok
Setting up trigger configuration: 
	 syncctrl forced_readout triggers: ok
ASIC 7
----------------
Setting config; response:  error: config_check
Setting up trigger configuration: 
	 syncctrl forced_readout triggers: ok
ASIC 8
----------------
Setting config; response:  error: config_check
Setting up trigger configuration: 
	 syncctrl forced_readout triggers: ok
ASIC 9
----------------
Setting config; response:  error: config_check
Setting up trigger configuration: 
	 syncctrl forced_readout triggers: ok
ASIC 10
----------------
Setting config; response:  error: config_check
Setting up trigger configuration: 
	 syncctrl forced_readout triggers: ok
ASIC 11
----------------
Setting config; response:  ok
Setting up trigger configuration:

adamschoenwald, 2020-12-01:
Prior to swap:
'''ASIC's 0,1,2,3 are working.
ASIC's 4,5 can clock config, but data is all 0
ASIC's 6-11 (side B) are not clocking configs.'''

adamschoenwald, 2020-12-01:
Found the following problems in the schematics:
Schematic | actual
a6o5 - a7o6
a6o6 - a7o5
a7o5 - a6o6
a7o6 - a6o5

a8o5 - a9o6
a8o6 - a9o5
a9o5 - a8o6
a9o6 - a8o5

a10o5 - a11o6
a10o6 - a11o5
a11o5 - a10o6
a11o6 - a10o5

SeanCGriffin, 2020-12-01:
This merge fixed the issue,

closed 2020-12-01 by SeanCGriffin