ComPair/ComPair-tracker-FPGA #148: O6 problems on some ASICS

adamschoenwald, 2021-01-29:


SeanCGriffin, 2021-01-29:
ASIC Vthr_DAC = 1 for these tests. Readout is disabled so should just be looking at triggers. 

AFE S/N 362/008: 
With probe: No trigger data on A5-06; data on A3-O6 but intermittent and not always in the same place (i.e noisy trigger??). Confirmed with scope that A5-06 has data during readout. 

[image]
[image]

SeanCGriffin, 2021-01-29:
AFE S/N 362/018:
With probe: Triggers on all ASICs; A5-06 is a BIT noisy.

Cranked threshold way up (31); see data on ALL channels during forced readout. 

Data not seen in web interface. 

[image]

SeanCGriffin, 2021-01-29:
Finally, AFE 362/018: A4, A5 do not show triggers in ILA when pulse is sent. 

Recall, in live display, 2 ASICs are missing data. I have swapped AFEs before and watched the missing data go to A4,A5 on Side A; I think there's a mapping issue somewhere in raw2hdf. 

[image]

Pulse was checked at attenuator input on scope; pulses visible for these ASICs.

adamschoenwald, 2021-02-09:
[image]
[image]

SeanCGriffin, 2021-02-12:
[image]

IC looks good. No shorts visible.

SeanCGriffin, 2021-02-12:
[image]
[image]

LVDS transmitter lines are pegged to 2.7 V; inputs to transmitter looks OK. 

No shorts to ground; LVDS pins are open when disconnected from DBE, 100 Ohm when connected.

SeanCGriffin, 2021-03-03:
Ken Simms eyeballed the chip, determined that Pin 1 wasn't soldered correctly. Data now present. Closing out the issue. 

[image]
[image]

closed 2021-03-03 by SeanCGriffin