ComPair/ComPair-tracker-FPGA #152: Something between the client and dacctrl/calctrl isn't working.

SeanCGriffin, 2021-02-04:
Sending dacctrl / calctrl commands doesn't work reliably (or at all sometimes).

SeanCGriffin, 2021-02-04:
e.g. the snippit below didn't turn on the DAC for Side A this time. 

```
client.dac_set_delay()

client.cal_pulse_stop()
DAC = 333
client.dac_set_counts("A", "vth", DAC)
client.dac_set_counts("B", "vth", DAC)
client.cal_settings(100, 100, 99900)
client.cal_pulse_start()
```

SeanCGriffin, 2021-02-10:
Looks like the problem lies in the connection to dacctrl.

SeanCGriffin, 2021-02-19:
Tested again today; sometimes `client.dac_set_counts()` works, sometimes it doesn't. Hard to make it happen but once it stops working usually doesn't come back.

lucasparker, 2021-03-16:
I couldn't find an obvious issue with the dac ctrl stuff. There was an issue with how the silayer server was accepting commands to modify the cal settings, which #167 should fix.

SeanCGriffin, 2021-03-18:
Noticed that only one side was working at a time and it was the SECOND side to be called in the following; this applied to both CAL and VTHR. 
```
client.dac_set_delay()
DAC = 1000
client.dac_set_counts("A", "CAL", DAC)
client.dac_set_counts("B", "CAL", DAC)
```

Inserting a delay between the two commands has solved the problem. 
```
client.dac_set_delay()
DAC = 1000
client.dac_set_counts("B", "CAL", DAC)
sleep(0.01)
client.dac_set_counts("A", "CAL", DAC)
```

Closing out issue, looks like we have a solution.

closed 2021-03-18 by SeanCGriffin