ComPair/ComPair-tracker-FPGA #154: AFE 362/008 : Cal pulses only seen in A1 and A4.

SeanCGriffin, 2021-02-09:
See plots in #148 . 

Verified with a probe that pulses can be seen with an attenuator and verified with microscope that the ASIC pin looks OK (not shorted to adjacent pins). ASICs in question all passed QC charge injection.

adamschoenwald, 2021-02-09:
From #148 before split, From Sean:
****
ASIC Vthr_DAC = 1 for these tests. Readout is disabled so should just be looking at triggers.

AFE S/N 362/008:
With probe: No trigger data on A5-06; data on A3-O6 but intermittent and not always in the same place (i.e noisy trigger??). Confirmed with scope that A5-06 has data during readout.

[image]

[image]

SeanCGriffin, 2021-02-09:
Correcting ILA plot: 

[image]

adamschoenwald, 2021-02-09:
Check that the VTHR pin is not shorted to +1.5 or some other random value. Should match what is set on the internal DAC if not being over-ridden.

[image]

adamschoenwald, 2021-02-09:
Alternate top side probe point [image]

SeanCGriffin, 2021-02-09:
[image]

SeanCGriffin, 2021-02-09:
Measured with a probe; voltages are a BIT higher than expected as per the datasheet but not at the voltage rail. 

e.g. Vthr_dac = 16 expects 96.6 mV, I measure about 110mV (on all channels). 
Vthr_dac = 32 expects 181.7 mV, measure about 205 mV

Checked for each ASIC on AFE008 and A0 on AFE018

adamschoenwald, 2021-02-11:
Measure the mbias point shown in this figure. The mbias current will be i_mbias = (1.5 - Vmeasure) / 48.7e3 Amps.
Compare this value on asics that are working properly vs ascis that are not triggering correctly.

[image]

adamschoenwald, 2021-02-11:
Also verify the asics are actually in state 3 ( [s2,s1,s0] = [0,1,0]) by probing each line w/ the oscilloscope dc coupled. Expect +1.5V for a 1, -2.0V for a 0.
[image]
[image]

SeanCGriffin, 2021-02-11:
I'll have a look. I think the states have to be working because data and
config readout works but this is an easy check.

SeanCGriffin, 2021-02-12:
Confirmed ASICs in M3.

adamschoenwald, 2021-02-12:
If S1 was stuck on 0, it would let you do the config regs but not go into modes that produce triggers.

SeanCGriffin, 2021-02-16:
[image]
[image]

Probed ASIC pins, triggered off of opposite side's cal pulse (hence the reflection)

SeanCGriffin, 2021-03-18:
This is weird -- changing the test channel seems to have brought an ASIC back??

test_channel = 10:
[image]

test_channel = 18: 
[image]

test_channel = 0: 
[image]