ComPair/ComPair-tracker-FPGA #166: AFE 362/015: TP8 short to ground while in use.

SeanCGriffin, 2021-03-04:


SeanCGriffin, 2021-03-04:
Testing mbias probe points: 

[image]
[image]

- A0: 51.70k
- A1: 51.70k
- A2: 51.70k
- A3: 51.70k
- A4: 51.70k
- A5: 51.70k

adamschoenwald, 2021-03-05:
We could check the voltage of the Mbias lines while the AFE is on to see if there are any outliers.

I think it might make more sense to try viewing this with a thermal cam as it starts to heat up to see if a shorted component is obvious.

SeanCGriffin, 2021-03-05:
Test configuration: ONLY power cable installed. AFE on DBE 003 Side A.  

Standard fee_enable: 
- -2.4V trips OCP @ 0.7 A
- +1.9 goes to 0.25A for about a second and then drops to 0.005A. 

Tried ramping up power on -2.4V line: 
- -2.4  200mA current draw when set to 1.6 V, trips OCP.

SeanCGriffin, 2021-03-05:
Re-measured mbias "test point" across 47.7kOhm resistor only at TP8 (analogue test point) 

- 48.7kOhm
- 48.7kOhm
- 48.73kOhm
- 48.8kOhm
- 48.8kOhm
- 48.77kOhm

SeanCGriffin, 2021-03-05:
fee enabled; -2.4 V off. 

All ASIC test points are at ~0.65 V.